don/NDEF #22: NFC Share

armaldio, 2015-03-31:
Hello don, 
I explain my situation : 

I have a Yùn board with nfc seeedstudio plug on it.
I have a Sony Xperia SP as mobile.

I load the nfc arduino program called "P2P receive" on the arduino board.
I wrote a cordova app that use this library : https://github.com/chariotsolutions/phonegap-nfc

Then, i share content to the arduino via my app (share, not write)
And i got results.

But i cannot see what i sent from my phone.
 I get two NDEF messages, one thet say it came from my phone with the package name and another that link to the play store to my app and it is concatenate with "&beam" or something like that.
(i'm don't have the board right niw with me)

Any help or suggestions ?
I hope i was as clear as possible ^^

Thank you very much in advance.

don, 2015-04-02:
It sounds like the Arduino part of your code is working fine. Those 2 NFC records you're seeing are the default records that an Android apps shares via beam. Basically it's a link to the Play store. Normally Android will disable this when you're sharing a NFC record from your app. I think you have a problems with the Cordova app. 

Try using this sample project https://github.com/don/nfc-share to get started with phonegap-nfc and sharing tags. If the default code sends a URI correctly, you can modify the code to send a Text Record

Replace https://github.com/don/nfc-share/blob/master/www/js/index.js#L41

    nfc.share([ ndef.uriRecord("http://cordova.io") ], success, failure);

with 

    nfc.share([ ndef.textRecord("Hello Arduino!") ], success, failure);

armaldio, 2015-04-02:
Hello again.
I just test you code.
it seems to your well with uri record

```
NDEF Message 1 record, 15 bytes
  NDEF Record
    TNF 0x1 Well Known
    Type Length 0x1 1
    Payload Length 0xB 11
    Type 55  U
    Payload 03 63 6F 72 64 6F 76 61 2E 69 6F  .cordova.io
    Record is 15 bytes
```

but when textrecord is occured
```
NDEF Message 2 records, 108 bytes
  NDEF Record
    TNF 0x1 Well Known
    Type Length 0x1 1
    Payload Length 0x45 69
    Type 55  U
    Payload 03 70 6C 61 79 2E 67 6F 6F 67 6C 65 2E 63 6F 6D 2F 73 74 6F 72 65 2F 61 70 70 73 2F 64 65 74 61 69 6C 73 3F 69 64 3D 6E 6F 74 2E 79 65 74 2E 73 70 65 63 69 66 69 65 64 26 66 65 61 74 75 72 65 3D 62 65 61 6D  .play.google.com/store/apps/details?id=not.yet.specified&feature=beam
    Record is 73 bytes
  NDEF Record
    TNF 0x4 External
    Type Length 0xF 15
    Payload Length 0x11 17
    Type 61 6E 64 72 6F 69 64 2E 63 6F 6D 3A 70 6B 67  android.com:pkg
    Payload 6E 6F 74 2E 79 65 74 2E 73 70 65 63 69 66 69 65 64  not.yet.specified
    Record is 35 bytes
```

Thank you.

armaldio, 2015-04-02:
Is it possible to get only the payload ?
I'm not able to convert NdefMessage to String.

don, 2015-04-06:
Your second example for the text record looks like the default NFC share generated by Android, which likely means that you have an error in your Cordova code. Try running `jshint` on index.js.

don, 2015-04-06:
To get the payload as a String try something like https://github.com/don/NDEF/blob/master/examples/ReadTagExtended/ReadTagExtended.ino#L70

You'll need to do more work if you have non-ascii characters in your string.

armaldio, 2015-04-07:
All is working, thank you very much :)

closed 2015-04-07 by armaldio