don/NDEF #23: Launch an app

armaldio, 2015-03-31:
Is it possible to launch an app while scanning the nfc shield ?
(Like an AAR Android Application Record)

xiongyihui, 2015-04-01:
Yes, just add an AAR record to your NDEF message.

```
const uint8_t aarType[] = "android.com:pkg";
const uint8_t aarPayload[] = "com.seeedstudio.nfcdemo";    // Your app package name
NdefRecord aarRecord = NdefRecord();
NdefMessage message = NdefMessage();

aarRecord.setTnf(TNF_EXTERNAL_TYPE);
aarRecord.setType(aarType, sizeof(aarType) - 1);
aarRecord.setPayload(aarPayload, sizeof(aarPayload) - 1);
message.addRecord(aarRecord);
```

armaldio, 2015-04-01:
Oh thank you very much.
i'll try that tomorow :)

don, 2015-04-02:
Thanks @xiongyihui

closed 2015-04-02 by don

armaldio, 2015-04-02:
Works like a charm ;) Ty again.