don/NDEF #24: Need help understanding - Adafruit PN532 Breakout Board

izkara, 2015-06-30:
I see in the readme that it should work, but I am constantly getting the "Send a message to Peer
Failed" print. I am using the Adafruit PN532 Breakout walk-through to configure the wiring for SPI, and it works using the Adafruit library. However in order to get P2P working with Android, this library seems like my best option if I can just figure out why it's failing.

Can you please provide an explanation for what this line of code means;

don, 2015-06-30:
My code uses the [Seeed Studio PN532 driver](https://github.com/Seeed-Studio/PN532). I only use I2C with the Adafruit shield and SPI with the Seeed shield. Theoretically the Adafruit shield should work fine SPI if you convert it over. I suspect the Adafruit NFC breakout board could work too, but you'll need to wire it for the Seeed library. Typically SS, MISO, MOSI and SCK are 10, 11, 12, 13. I think you'll need to adjust the wiring (or code) to use the Seeed driver with the breakout board.

If you do get this working, please post or email the details so I can update the documentation for others.

izkara, 2015-07-01:
I did get it working after a simple re-wiring, thank you! From the original wiring in the Adafruit tutorial I took the pins in my Arduino Uno 2, 3, 4, 5 and moved them over to 13, 11, 10, 12 respectively.  I've also connected the breakout board's IRQ to Arduino's pin 2, just in case. I kept the level shifter as well, so the only difference between mine and the Adafruit Breakout wiring are those small changes.

Although I certainly got the PN532 to push messages to my phone via NFC now using the basic P2P Send example, it is unfortunately unsuccessful running P2P Receive. I simply get this message: 

> NFC Peer to Peer Example - Receive Message
> Waiting for message from Peer
> Failed

I am using the StickyNote sample for Android. Do you recommend any apps, or perhaps one I can work off of from Beginning NFC, that can push and pull text with the PN532?

don, 2015-07-01:
I use Chrome on Android for testing P2P. Chrome sends the URL current page via NFC. If Android receives a Well Known message with RTD URL, it will open Chrome.

izkara, 2015-07-09:
Don-

Thanks for the help. I did manage to get communication going both ways. However the only way I can get a phone to send data to the reader is by using Android Beam, in which I have to tap on the device to confirm a sending. Is there a way to send data without Beam, and just have it auto-send as soon as it detects the reader?

don, 2015-10-08:
@izkara Unfortunately Beam is the way Android likes to send data, but it's pretty clunky. I've heard that HCE (Host-based Card Emulation) might be able to send data without the "push to beam" prompt but I haven't had a chance to use it yet.

closed 2015-10-08 by don

aeyuxi, 2017-12-28:
@don I just hit this issue. I was wondering if there is a way of changing the pins in the code. The only part where I see that the pins are changed defined is is when PN532_SPI pn532spi(SPI, 10) is defined.