don/NDEF #26: NDEF with Adafruit_NFCShield_I2C

neo4700, 2015-07-20:
HI
kindly, i have arduino DUE, and Adafruit PN532 NFC/RFID Shield, and the only library which is compatible (as i heard) is the Adafruit_NFCShield_I2C library, the thing is that i want to use the NDEF library developed by you (interacting with records is easy), but as i heard that it needs some modification to work with Adafruit_NFCShield_I2C library, so can you please help me with that.

Note: the reason im using DUE is because of its large memory

neo4700, 2015-07-22:
Hello @don 
Kindly , would you plz help me in the above issue?

don, 2015-10-08:
@neo4700 Sorry for the very late response...

You can use this library with the Arduino Due and the Adafruit NFC Shield. You need to use the Seeed Studio PN532 drivers rather than the Adafruit drivers.

In order to use the Due, you need to jumper SDA and SCL on the Shield to pins to 20 and 21 on the Due. (Alternately pins 4 and 5 on the shield should work too see https://www.arduino.cc/en/Reference/Wire ) 

[image]

closed 2015-10-08 by don