don/NDEF #27: Send char more than 115 char over P2P

mazirul, 2015-08-28:
Don, can you help me??I want to send character/string over P2P to phone, ijust can send only 115 character with mimemediarecors. if I use TextRecord, it just can transfer 88 char.

How can I send A lot of string/character with P2P?maybe about 600 characters..

Thanks for the answer

Regards.
Khoirul

don, 2015-12-08:
@mazirul Sorry about the delayed response. I'm just seeing this now.

Theoretically you should be able to send 600 characters over P2P. I expect you're running into memory limitations.

In the past I've been able to write large messages to NFC tags by running this code on an Arduino Due. I'd expect the same thing would work for the peer-to-peer. 

It's possible there are some size limitation in the SNEP and LLCP code. You'd need to check the [Seeed Studio](https://github.com/Seeed-Studio/PN532) repo to verify.

don, 2015-12-09:
The maximum NDEF buffer I can write from the Arduino to Android is 233 bytes. Anything larger fails when using the Seeed Studio NFC Shield v2.0 and a Nexus 5X.

A 226 character string encoded into a TNF WELL_KNOWN RTD Text message is 233 bytes.

Encoding 226 character textRecord with NDEF library runs an Uno out of memory and fails. It works OK on an Arduino Due.

https://gist.github.com/don/2bf283f81485f89dca13#file-p2p_send_longstring-ino

Writing the bytes into the message works OK on an Uno.

https://gist.github.com/don/2bf283f81485f89dca13#file-p2p_send_longrecord-ino

It looks like this is a limit in the hardware or PN532 library. I eliminated the NDEF library and sent a 226 character string in a text record as raw NDEF bytes using only the PN532 code.

https://gist.github.com/don/2bf283f81485f89dca13#file-p2p_send_raw-ino

You can use [this script](https://gist.github.com/don/2bf283f81485f89dca13#file-generate_ndef_bytes-js) to generate NDEF messages for testing.

closed 2015-12-09 by don

don, 2016-02-17:
See #32