don/NDEF #29: Get message as a String

ben14124, 2015-10-08:
Hi, is it there any form to get the message that is in the NFC card and store it in a String variable in Arduino?

Thanks!

don, 2015-10-08:
@ben14124 the payload comes back as bytes. You can create a string from the bytes.

If you read a TNF_Well_Known tag with RTD of Text, you'd read the first byte to get the length of the language code, then read the language code, then convert the remaining bytes into a string.

The [ReadTagExtended.ino example](https://github.com/don/NDEF/blob/master/examples/ReadTagExtended/ReadTagExtended.ino#L68-L75) attempts to convert every NDEF message payload into a String. This works for OK some tags and fails miserably for others.

        String payloadAsString = "";
        for (int c = 0; c < payloadLength; c++) {
          payloadAsString += (char)payload[c];
        }

ben14124, 2015-10-08:
Oh yeah, I totally forgot the ReadTagExtended example. That worked perfectly for me. Thanks a lot!

closed 2015-10-08 by don