don/NDEF #32: P2P_Sending limitation

lorant124, 2016-02-14:
Hi Don,

I would like to ask for help, I m currently using your library of NFC to  send messages to android over p2p protocol. I would like to send at least 2000 characters. 

Is It possible or it is given by the hardrware PN532, I use ESP8266 and  now I'm able to send 223 byte.  I know that the library limites the number of bytes "SNEP packet, the packet should be less than (255 - 2 - 3) bytes". My question is if it was possible to send over this limit?

Thank you very much for your answear and your help,

Lorant

don, 2016-02-17:
The NDEF library can encode messages of 2000 characters. Unfortunately it looks like the SNEP write function in PN532 is limited to 250 bytes https://github.com/Seeed-Studio/PN532/blob/master/PN532/snep.h#L23

I don't know why this limit exists and if it can be changed. I've emailed @xiongyihui  to see if he has more info. Maybe large messages need to call snep::write multiple times?

diegosilvars, 2016-10-14:
Hello. any news?