don/NDEF #36: NFC ans Samsung

feelgoodgirl, 2017-05-10:
Hey,
In your description, I saw that this library should  good for my project. My goal is to read and send messages with a smartphone with NFC.

When I introduced your library and chanch the follinging in the code:
```
#include <Wire.h>
#include <PN532_I2C.h>
#include <PN532.h>
#include <NfcAdapter.h>

PN532_I2C pn532_i2c(Wire);
NfcAdapter nfc = NfcAdapter(pn532_i2c);
```
i got the measege that  #include <PN532.h> where not include. 
After I added this library, I received the following message:
```
Arduino: 1.8.2 (Windows 7), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from C:\Users\364\AppData\Local\Temp\arduino_modified_sketch_533945\P2P_Send.ino:6:0:

C:\Users\364\Documents\Arduino\libraries\PN532/PN532.h:112:7: error: redefinition of 'class PN532'

 class PN532

       ^

In file included from C:\Users\364\Documents\Arduino\libraries\arduino_NFC-master/PN532_I2C.h:15:0,

                 from C:\Users\364\AppData\Local\Temp\arduino_modified_sketch_533945\P2P_Send.ino:5:

C:\Users\364\Documents\Arduino\libraries\arduino_NFC-master/PN532_Com.h:74:7: error: previous definition of 'class PN532'

 class PN532{

       ^

P2P_Send:9: error: no matching function for call to 'PN532_I2C::PN532_I2C(TwoWire&)'

 PN532_I2C pn532_i2c(Wire);

                         ^

C:\Users\364\AppData\Local\Temp\arduino_modified_sketch_533945\P2P_Send.ino:9:25: note: candidates are:

In file included from C:\Users\364\AppData\Local\Temp\arduino_modified_sketch_533945\P2P_Send.ino:5:0:

C:\Users\364\Documents\Arduino\libraries\arduino_NFC-master/PN532_I2C.h:26:5: note: PN532_I2C::PN532_I2C(uint8_t, uint8_t)

     PN532_I2C(uint8_t irq, uint8_t reset);

     ^

C:\Users\364\Documents\Arduino\libraries\arduino_NFC-master/PN532_I2C.h:26:5: note:   candidate expects 2 arguments, 1 provided

C:\Users\364\Documents\Arduino\libraries\arduino_NFC-master/PN532_I2C.h:24:7: note: constexpr PN532_I2C::PN532_I2C(const PN532_I2C&)

 class PN532_I2C : public PN532{

       ^

C:\Users\364\Documents\Arduino\libraries\arduino_NFC-master/PN532_I2C.h:24:7: note:   no known conversion for argument 1 from 'TwoWire' to 'const PN532_I2C&'

C:\Users\364\Documents\Arduino\libraries\arduino_NFC-master/PN532_I2C.h:24:7: note: constexpr PN532_I2C::PN532_I2C(PN532_I2C&&)

C:\Users\364\Documents\Arduino\libraries\arduino_NFC-master/PN532_I2C.h:24:7: note:   no known conversion for argument 1 from 'TwoWire' to 'PN532_I2C&&'

P2P_Send:10: error: no matching function for call to 'NfcAdapter::NfcAdapter(PN532_I2C&)'

 NfcAdapter nfc = NfcAdapter(pn532_i2c);

                                      ^

C:\Users\364\AppData\Local\Temp\arduino_modified_sketch_533945\P2P_Send.ino:10:38: note: candidates are:

In file included from C:\Users\364\AppData\Local\Temp\arduino_modified_sketch_533945\P2P_Send.ino:7:0:

C:\Users\364\Documents\Arduino\libraries\NDEF-master/NfcAdapter.h:25:9: note: NfcAdapter::NfcAdapter(PN532Interface&)

         NfcAdapter(PN532Interface &interface);

         ^

C:\Users\364\Documents\Arduino\libraries\NDEF-master/NfcAdapter.h:25:9: note:   no known conversion for argument 1 from 'PN532_I2C' to 'PN532Interface&'

C:\Users\364\Documents\Arduino\libraries\NDEF-master/NfcAdapter.h:23:7: note: constexpr NfcAdapter::NfcAdapter(const NfcAdapter&)

 class NfcAdapter {

       ^

C:\Users\364\Documents\Arduino\libraries\NDEF-master/NfcAdapter.h:23:7: note:   no known conversion for argument 1 from 'PN532_I2C' to 'const NfcAdapter&'

C:\Users\364\AppData\Local\Temp\arduino_modified_sketch_533945\P2P_Send.ino: In function 'void loop()':

P2P_Send:35: error: no matching function for call to 'NfcAdapter::write(uint8_t [128], int&)'

     if (0 >= nfc.write(ndefBuf, messageSize)) {

                                            ^

C:\Users\364\AppData\Local\Temp\arduino_modified_sketch_533945\P2P_Send.ino:35:44: note: candidate is:

In file included from C:\Users\364\AppData\Local\Temp\arduino_modified_sketch_533945\P2P_Send.ino:7:0:

C:\Users\364\Documents\Arduino\libraries\NDEF-master/NfcAdapter.h:31:17: note: boolean NfcAdapter::write(NdefMessage&)

         boolean write(NdefMessage& ndefMessage);

                 ^

C:\Users\364\Documents\Arduino\libraries\NDEF-master/NfcAdapter.h:31:17: note:   candidate expects 1 argument, 2 provided

exit status 1
no matching function for call to 'PN532_I2C::PN532_I2C(TwoWire&)'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

Could you possibly help me?

flosommerfeld, 2017-06-16:
I have the same problem

don, 2018-05-12:
It sounds like PN532 libraries might not be installed correctly. This library is a pain to install because of the multiple libraries and lack of Arduino IDE integration.

Here's how I install it on my Mac

    cd ~/Documents/Arduino/libraries
    git clone https://github.com/don/NDEF.git
    git clone https://github.com/Seeed-Studio/PN532 NFC
    ln -s NFC/PN532 .
    ln -s NFC/PN532_SPI .
    ln -s NFC/PN532_I2C .
    ln -s NFC/PN532_HSU .

don, 2018-05-12:
On Windows it's similar but I copy the directories rather than creating symbolic links

Open a cmd prompt

    cd Documents\Arduino\libraries
    git clone https://github.com/don/NDEF.git
    git clone https://github.com/Seeed-Studio/PN532 NFC
    move NFC\PN532 .
    move NFC\PN532_HSU .
    move NFC\PN532_I2C .
    move NFC\PN532_SPI .
    rd /s /q NFC

Hopefully that helps. This issue is old so I'm closing it. Open a new issue if you're still having a problem.

closed 2018-05-12 by don