don/NDEF #44: How to use NDEF_USE_SERIAL and NDEF_DEBUG defines in order to debug code

josebagarciamiranda, 2018-06-10:
Hello, everyone:

I know this could be a silly question but I would like to know how to use NDEF_USE_SERIAL and NDEF_DEBUG defines in order to debug the obtained data in the _NdefRecord.cpp_ and _NdefMessages.cpp_  code.

I tried to insert both defines in the library code and the arduino code but I do not get extra serial outputs for debugging in the arduino serial monitor. I do not even know if the serial output of the library should be watch here...

I am using NDEF library with PN532 library from seed studio in order to use my NFC PN532 board to communicate with my phones.

Thank you very much in advance.

don, 2018-06-13:
Not a silly question at all. I can't figure out the Arduino preprocessor. O_o  Unfortunately adding a #define in the Arduino sketch won't have any effect on the library. That's how I'd like it to work.

You need to modify the library code. The #defines need to be added to odd places to get them to work. Try:

Disable NDEF_USE_SERIAL by commenting out line 5 in Ndef.h 
https://github.com/don/NDEF/blob/838d3b68a62c5b95158b4ab4bdeabb3a5225001f/Ndef.h#L5

Disable Mifare Support by commenting out line 5 in Mifare.h
https://github.com/don/NDEF/blob/838d3b68a62c5b95158b4ab4bdeabb3a5225001f/MifareClassic.h#L5

I haven't used NDEF_DEBUG in a long time I can't remember where I defined that. Try to add a define in NfcAdapter.h.

IIRC, NDEF_USE_SERIAL should be enabled by default and NDEF_DEBUG is disabled. Serial output needs to be on for the [example code](https://github.com/don/NDEF/tree/master/examples) to work but this library barely has any memory running on an Uno so disabling stuff is a good idea. Note that the current code was updated recently by merging some ancient pull requests. I haven't used it much.

If you have suggestions on how to make this better please let me know.

rrakso, 2020-10-29:
@josebagarciamiranda for me it worked only if I put these defines [image] 
above the project libraries in my main project file 😉