don/NDEF #53: NdefMessage addTextRecord maxing out.

tgikal, 2019-06-24:
I'm trying to write to a NTAG215, which has around 4K of memory, yet I can only seem to use 227 bytes with the `message.addTextRecord(input)` function.

Basically I have something that looks like this:

```
#include <SPI.h>
#include <PN532_SPI.h>
#include <PN532.h>   // The following files are included in the libraries Installed
#include <NfcAdapter.h>

PN532_SPI pn532spi(SPI, 10);
NfcAdapter nfc = NfcAdapter(pn532spi);  // Indicates the Shield you are using
String input;

void setup(void) {
  while (!Serial); // for Leonardo/Micro/Zero
  Serial.begin(9600);
  Serial.println("READY"); // Header used when using the serial monitor
  nfc.begin();
}
void loop(void) {
  while(Serial.available() > 1) {
    input = Serial.readString(); } // read the incoming data as string

  if (input.indexOf("write") >= 0) {
    if (nfc.tagPresent()) {
      NdefMessage message = NdefMessage();
      input.remove(0,6);
      message.addTextRecord(input); // Text Message you want to Record
      boolean success = nfc.write(message);
      if (success) {
          Serial.println("COMPLETE"); // if it works you will see this message 
      } else {
          Serial.println("FAILED"); } // If the the rewrite failed you will see this message
    }
  } // End of write
  while (nfc.tagPresent()); // Wait for tag removal
  input = "";
}
```

Which works fine, until I exceed 227 characters.

Any ideas?

tgikal, 2019-06-25:
Looking around it looks like declaring `input` as a `String` is going to cause all kinds of memory management issues since `String` is apparently terribly with memory.
Could the issue be related to this?
And will `message.addTextRecord` accept a char array?

tgikal, 2019-06-25:
I found it in the MifareUltralight.cpp, `#define ULTRALIGHT_MAX_PAGE 63` was limiting the tag to 63 pages (~252 bytes), I changed it to  `#define ULTRALIGHT_MAX_PAGE 135` and it seems to be working.

tgikal, 2019-06-26:
That actually only bumped it up to 260 characters before something else is maxing out, basically it crashes the arduino is you go higher.

rrakso, 2020-10-29:
@tgikal hmm, [this issue](https://github.com/don/NDEF/issues/60) (#60) seem to be alittle bit similar. I will try to figure something out.